derolli1976/enpal #1: Sensoren werden nicht im Energie Dashboard Dropdown angezeigt

DevRealHard, 2025-05-07:
Die Sensoren aus der Integration werden nicht im Dropdown bei der Konfiguration des Engergy Dashboards angezeigt.
Laut https://www.home-assistant.io/docs/energy/faq/#troubleshooting-missing-entities müssen die Device_class = power sein und die state_class = total oder total_increasing. 

Die device_class stimmt bei den in Frage kommenden Sensoren, nur die state_class ist nicht spezifiziert und daher stehen die Sensoren im Energy Dashboard nicht zur Verfügung.

derolli1976, 2025-05-07:
Nur kurz, damit das hier korrekt steht.

Device Class für die "Wh" und "kWh" Sensoren für das Energiedashboard muss "energy" sein. Es fehlt aber die state_class. Korrekt?

Folcr, 2025-05-09:
Hallo Zusammen, also enn device_class = energy und state_class= totel, total._increasing wäre, kann man das im Energy Dashboard einfach auswählen. Zur Zeit ist die device_class = power.

[image]

Wenn das noch geändert würde, ein Träumchen! Vielen Dank. Tolle Arbeit.

derolli1976, 2025-05-09:
Ist im 2.0 Release gefixt. Sollte in den nächsten Tagen releasefertig sein.
Ihr könnt aber gerne das Beta Release installieren.

closed 2025-05-10 by derolli1976