derolli1976/enpal #10: [Bug] Translation error: UNCLOSED_TAG

WILHER, 2025-05-11:
### Version der Integration

2.0.0

### Home Assistant Version

2025.5.1

### Installationsart von Home Assistant

Home Assistant OS

### Was ist das Problem?

Wenn ich die Wallbox Steuerung aktivieren will kommt die Fehlermeldung: Translation error: UNCLOSED_TAG

### Schritte zur Reproduktion

Nach Installation der Integration und des Add-ON ist dieser Fehler bei mir dauerhaft da.

### Home Assistant Logausgaben

```shell

```

### Relevanter Konfigurationsauszug

```yaml

```

### Verwendest du das Wallbox Add-on?

Ja

### Version des Wallbox Add-ons

1.0.5

### Logausgaben des Wallbox Add-ons

```shell
2025-05-11 17:23:22,952 [DEBUG] DELETE http://localhost:39117/session/7f26263ae4d1d0bf8771b97ce34c6131 {}
2025-05-11 17:23:23,005 [DEBUG] http://localhost:39117 "DELETE /session/7f26263ae4d1d0bf8771b97ce34c6131 HTTP/1.1" 200 0
2025-05-11 17:23:23,005 [DEBUG] Remote response: status=200 | data={"value":null} | headers=HTTPHeaderDict({'Content-Length': '14', 'Content-Type': 'application/json; charset=utf-8', 'cache-control': 'no-cache'})
2025-05-11 17:23:23,005 [DEBUG] Finished Request
2025-05-11 17:23:23,015 [INFO] 172.30.32.1 - - [11/May/2025 17:23:23] "GET /wallbox/status HTTP/1.1" 200 -
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: Enpal Wallbox Controller
 Steuerung der Enpal Wallbox über Buttons via Selenium
-----------------------------------------------------------
 Add-on version: 1.0.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2025.5.1
 Home Assistant Supervisor: 2025.04.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2025-05-11 17:25:28,063 [INFO] Options loaded successfully.
2025-05-11 17:25:28,067 [INFO] Starting Wallbox API service on port 36725...
 * Serving Flask app 'run'
 * Debug mode: off
2025-05-11 17:25:28,071 [INFO] WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:36725
 * Running on http://172.30.33.1:36725
2025-05-11 17:25:28,072 [INFO] Press CTRL+C to quit
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: Enpal Wallbox Controller
 Steuerung der Enpal Wallbox über Buttons via Selenium
-----------------------------------------------------------
 Add-on version: 1.0.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.2  (amd64 / generic-x86-64)
 Home Assistant Core: 2025.5.1
 Home Assistant Supervisor: 2025.04.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2025-05-11 17:30:54,803 [INFO] Options loaded successfully.
2025-05-11 17:30:54,807 [INFO] Starting Wallbox API service on port 36725...
 * Serving Flask app 'run'
 * Debug mode: off
2025-05-11 17:30:54,812 [INFO] WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:36725
 * Running on http://172.30.33.1:36725
2025-05-11 17:30:54,812 [INFO] Press CTRL+C to quit
```

WILHER, 2025-05-11:
Nach Deinstallation, System neustart und erneuer Add-ON Installation lässt es sich nun starten.
Ist also erledogt.

closed 2025-05-11 by WILHER

derolli1976, 2025-05-11:
War auch mit Version 2.0.1 behoben. Danke für die Meldung!