derolli1976/enpal #11: [Feature] Websocket verwenden

Trustmania, 2025-05-11:
### Dein Vorschlag

Juten Tag,

erstmal danke für die Integration. Gefällt mir sehr gut, besonders die Idee mit der WallBox Integration!
Ich habe auch schon mich mal ran gemacht ein bisschen mit der Enpal Seite rumzuspielen. Mir ist aufgefallen, dass es wohl einen Websocket gibt, welcher vom Frontend genutzt wird. Evtl. wäre es eine coole Lösung statt alle 60s zu pullen sich mit dem Websocket zu verbinden und so die Daten direkt zu bekommen.

Beim Reverse Engineering bin ich aus Mangel der Zeit noch nicht weiter gekommen.

### Betrifft dein Wunsch die Wallbox-Steuerung?

None

derolli1976, 2025-05-15:
Klar, gerne!
Wenn du da eine Lösung hast, können wir mal schauen, wie wir das integrieren. Danke!

arigon, 2025-12-25:
Ich hab mal den blazor websocket kram reverse engineered und eine Go library gebaut: https://github.com/arigon/enpal_data_collector

Es simuliert den click auf den rechten button: Load Current Collector State
[image]

derolli1976, 2025-12-25:
Hi! Vielen Dank!
Ich habe mir das einmal ganz oberflächlich angesehen. GO ist natürlich etwas ungünstig im Homeassistant Umfeld.
Die Basis ist aber extrem gut. Ich würde das mal in Python nachbauen und dann als neue Datenquelle integrieren.
Mal schauen, wie ich Zeit finde. Das wird dann ein riesen Sprung. Hoffen wir mal, dass sie an dem Framework festhalten.
Plan: Version 3.0. Das muss eine neue Major werden.

derolli1976, 2025-12-25:
Danke fürs Teilen. Ich schaue mir das vor allem auch im Bezug auf die Wallbox an. Dann kann ich den zweigleisigen Weg mit Extra Addon sparen.

Trustmania, 2025-12-26:
Kann man das nicht ergänzend machen? Bzw. ich würde davon ausgehen, dass es nur eine Frage von "anderen Funktionen" welche ausgeführt werden müssen um alle die gleichen Werte zu bekommen?!

@arigon coole Arbeit!

@derolli1976 echt cool, dass du da gleich drauf springst.

Euch allen frohe Weihnachten!

Trustmania, 2025-12-26:
Echt coole Übersicht. Ich hab gerade auch mal Wireshark angeschmissen und gucke auch nochmal tiefer rein. Ich bin davon überzeugt, dass die Werte die in der HTML angezeigt werden auch aus Funktionen aus dem Websocket kommen. Es ergibt in meinen Augen keinen Sinn die Daten über mehrere Wege zu beschaffen. Besonders weil sich die Daten ja auch aktualisieren und man sieht im Log, dass über den Websocket regelmäßig etwas kommt.

Ich gucke mal was ich rausfinde. Im besten Fall haben wir dadurch sogar eine Lösung die 1:1 das abbildet was wir aktuell via HTML sehen / bekommen + einiges mehr, weil enpal das für die Endkunden nicht visualisiert.

Trustmania, 2025-12-28:
> > Die ganze Kommunikation läuft über Websocket. Jeder Funktionsaufruf um eine HTML page zu rendern läuft darüber ab. Also sollten auch die Inhalte der anderen Seiten darüber abrufbar sein. Eine Sache noch zum Data Collector: der wird serverseitig nur alle 20 Sekunden aktualisiert, also nur alle 20s gibts eine neue collection-id. Das ist leider viel zu träge um Spannungsspitzen oder temporäre höhere Ladeströme zu erkennen.
> 
> Die 20 Sekunden scheinen tatsächlich so gesetzt zu sein, aber man kann durch "Run Collection Cycle" ein Update erzwingen. Es gibt auf jeden Fall dann eine neue collection_id. Inwieweit die Daten dann auch aktueller sind, kann ich so nicht sagen. Ein realtime Datenabruf scheint mir über diesen Weg jedenfalls nicht wirklich möglich. Das ließe sich vermutlich nur via Modbus direkt aus den Wechselrichtern bewerkstelligen. Es gibt insgesamt viel zu viele Latenzen.
> 
> Für den "Standard-User" sollte der 20 Sekunden Cycle aber wohl ausreichend sein. Im Homeassistant Umfeld ist das ohnehin nicht sinnvoll die Aktualisierungsrate so massiv zu erhöhen. Das ist auch keine Realtimeanwendung.

Naja es gibt ja drei Seiten wo man Daten bekommt. Der Run Collection Cycle scheint eher so eine einmalige "Debug" Geschichte für die Enpal Mitarbeiter zu sein.

Auf der Messages bzw. Eigenschaften Seite ist es auch so, dass die Websocket Verbindung genutzt wird und da scheinen sich die Werte alle 5s upzudaten. Da kommen dann nur die Änderungen via Websocket (so wie es sein soll).

Ich bin dabei uns eine Python-Lib zu bauen die ALLES lesen / verwenden kann. Also alle Informationen von allen Seiten und auch die Buttons also Collection und Wallbox.

derolli1976, 2025-12-28:
Mega! Dann kann ich schon einmal beginnen die Datenbeschaffung konfigurierbar zu machen. Würde das Parsing der Seite immer noch als Fallback beibehalten.
Auf jeden Fall ein riesen Schritt nach vorne! Auch wegen der Wallbox. Das vereinfacht das deutlich.

Trustmania, 2025-12-28:
Ich kann aber noch nicht versprechen, wann ich durch bin. Blazor hat einige Anforderungen die Nachrichten zu beantworten, besonders wenn die automatisch gesendet werden.