derolli1976/enpal #16: [Feature] Wallbox als On/Off Switch

Trustmania, 2025-05-13:
### Dein Vorschlag

Juten Morgen, 

wie schon mehrmals geschrieben, nochmal vielen Dank für den ganzen drive, die du in die Geschichte hier bringst. Extrem gut!

Zur Feature Idee: Wäre es nicht sinnvoll, die Wallbox als Switch abzubilden? Der "Stateless" Button ist erstmal ok für die erste Abbildung, weil wir generell die Möglichkeit bekommen haben, die Wallbox in HA einzubinden, ich denke aber einen Switch, der den Status auch abbildet wäre genialer, besonders auch für Automatisierungen.

So kann man sich einfache Automatisierungen machen wie z.b.:

WENN wallbox == on UND PV < 3kW UND Speicher < 80% DANN wallbox = off

### Betrifft dein Wunsch die Wallbox-Steuerung?

None

derolli1976, 2025-05-13:
Gute Idee! Schaue ich mir an.
Persönlich ermittle ich den Status "Lädt / Lädt nicht" aus der Energie, die die Wallbox gerade verbraucht. Mal schauen, welche Variante ich nehme.

Trustmania, 2025-05-13:
Ja, so habe ich es bisher auch gemacht, dass ich gucke ob > 1000w auf der WallBox anliegen.

commit 2119016d (2025-05-14): Merge pull request #18 from derolli1976/feature/16-feature-wallbox-als-onoff-switch

Added select box for lademodus and switch to turn on/off (Closes #16)
closed 2025-05-14 by derolli1976