derolli1976/enpal #24: [Bug] Liest die Daten nicht vom Wallbox addon

Daddycool2708, 2025-05-15:
### Version der Integration

V2.1.1

### Home Assistant Version

2025.5.1

### Installationsart von Home Assistant

Home Assistant OS

### Was ist das Problem?

Es hat den Anschein das er die Daten nicht lesen kann. Wenn ich im Protokoll vom Add-on schaue, seht das das er die Daten erfolgreich gelesen hat. Wenn ich manuell auf die URL gehe dann sehe ich auch was er gelesen hat von der Wallbox. Ich habe das Gefühl das mein System etwas länger benötigt um die Daten bereit zu stellen. Also der Enpal webparser nicht lang genug wartet.

Hier mal das Protokoll vom Add-on:

> 2025-05-15 09:29:39,291 [INFO] Retrieving wallbox status...
> 2025-05-15 09:29:39,293 [DEBUG] Initializing Firefox WebDriver...
> 2025-05-15 09:29:39,294 [DEBUG] Skipping Selenium Manager; path to firefox driver specified in Service class: /usr/bin/geckodriver
> 2025-05-15 09:29:39,298 [DEBUG] Started executable: `/usr/bin/geckodriver` in a child process with pid: 5416 using 0 to output -3
> 2025-05-15 09:29:39,312 [DEBUG] POST http://localhost:36393/session {'capabilities': {'firstMatch': [{}], 'alwaysMatch': {'browserName': 'firefox', 'acceptInsecureCerts': True, 'moz:debuggerAddress': True, 'pageLoadStrategy': <PageLoadStrategy.normal: 'normal'>, 'moz:firefoxOptions': {'prefs': {'remote.active-protocols': 1}, 'args': ['--headless', '--width=1920', '--height=1080']}}}}
> 2025-05-15 09:29:39,315 [DEBUG] Starting new HTTP connection (1): localhost:36393
> 2025-05-15 09:30:09,812 [DEBUG] http://localhost:36393 "POST /session HTTP/1.1" 200 0
> 2025-05-15 09:30:09,814 [DEBUG] Remote response: status=200 | data={"value":{"sessionId":"07837b6d-e3b5-470d-9c92-e3caf4ce7b64","capabilities":{"acceptInsecureCerts":true,"browserName":"firefox","browserVersion":"128.10.0","moz:accessibilityChecks":false,"moz:buildID":"20250421121556","moz:geckodriverVersion":"0.34.0","moz:headless":true,"moz:platformVersion":"6.12.23-haos","moz:processID":5424,"moz:profile":"/tmp/rust_mozprofileoU1Zzk","moz:shutdownTimeout":60000,"moz:webdriverClick":true,"moz:windowless":false,"pageLoadStrategy":"normal","platformName":"linux","proxy":{},"setWindowRect":true,"strictFileInteractability":false,"timeouts":{"implicit":0,"pageLoad":300000,"script":30000},"unhandledPromptBehavior":"dismiss and notify","userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:128.0) Gecko/20100101 Firefox/128.0"}}} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '761', 'date': 'Thu, 15 May 2025 07:29:39 GMT'})
> 2025-05-15 09:30:09,816 [DEBUG] Finished Request
> 2025-05-15 09:30:09,816 [DEBUG] Firefox WebDriver initialized successfully.
> 2025-05-15 09:30:09,818 [DEBUG] POST http://localhost:36393/session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/url {'url': 'http://192.168.178.153/wallbox'}
> 2025-05-15 09:30:21,167 [DEBUG] http://localhost:36393 "POST /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/url HTTP/1.1" 200 0
> 2025-05-15 09:30:21,172 [DEBUG] Remote response: status=200 | data={"value":null} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '14', 'date': 'Thu, 15 May 2025 07:30:09 GMT'})
> 2025-05-15 09:30:21,174 [DEBUG] Finished Request
> 2025-05-15 09:30:21,175 [DEBUG] POST http://localhost:36393/session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element {'using': 'xpath', 'value': "//h6[contains(text(), 'Mode ')]"}
> 2025-05-15 09:30:21,265 [DEBUG] http://localhost:36393 "POST /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element HTTP/1.1" 404 0
> 2025-05-15 09:30:21,267 [DEBUG] Remote response: status=404 | data={"value":{"error":"no such element","message":"Unable to locate element: //h6[contains(text(), 'Mode ')]","stacktrace":"RemoteError@chrome://remote/content/shared/RemoteError.sys.mjs:8:8\nWebDriverError@chrome://remote/content/shared/webdriver/Errors.sys.mjs:193:5\nNoSuchElementError@chrome://remote/content/shared/webdriver/Errors.sys.mjs:511:5\ndom.find/</<@chrome://remote/content/shared/DOM.sys.mjs:136:16\n"}} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '415', 'date': 'Thu, 15 May 2025 07:30:21 GMT'})
> 2025-05-15 09:30:21,267 [DEBUG] Finished Request
> 2025-05-15 09:30:21,771 [DEBUG] POST http://localhost:36393/session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element {'using': 'xpath', 'value': "//h6[contains(text(), 'Mode ')]"}
> 2025-05-15 09:30:21,803 [DEBUG] http://localhost:36393 "POST /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element HTTP/1.1" 200 0
> 2025-05-15 09:30:21,804 [DEBUG] Remote response: status=200 | data={"value":{"element-6066-11e4-a52e-4f735466cecf":"f1e0e771-f9b8-4850-95ba-9dcd2df61f59"}} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '88', 'date': 'Thu, 15 May 2025 07:30:21 GMT'})
> 2025-05-15 09:30:21,806 [DEBUG] Finished Request
> 2025-05-15 09:30:21,808 [DEBUG] POST http://localhost:36393/session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element {'using': 'xpath', 'value': "//p[contains(text(), 'Status ')]"}
> 2025-05-15 09:30:21,826 [DEBUG] http://localhost:36393 "POST /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element HTTP/1.1" 200 0
> 2025-05-15 09:30:21,827 [DEBUG] Remote response: status=200 | data={"value":{"element-6066-11e4-a52e-4f735466cecf":"99f005e5-3332-46ee-a155-b726c9b56962"}} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '88', 'date': 'Thu, 15 May 2025 07:30:21 GMT'})
> 2025-05-15 09:30:21,828 [DEBUG] Finished Request
> 2025-05-15 09:30:21,829 [DEBUG] POST http://localhost:36393/session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element {'using': 'xpath', 'value': "//h6[contains(text(), 'Mode ')]"}
> 2025-05-15 09:30:21,884 [DEBUG] http://localhost:36393 "POST /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element HTTP/1.1" 200 0
> 2025-05-15 09:30:21,887 [DEBUG] Remote response: status=200 | data={"value":{"element-6066-11e4-a52e-4f735466cecf":"f1e0e771-f9b8-4850-95ba-9dcd2df61f59"}} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '88', 'date': 'Thu, 15 May 2025 07:30:21 GMT'})
> 2025-05-15 09:30:21,889 [DEBUG] Finished Request
> 2025-05-15 09:30:21,890 [DEBUG] GET http://localhost:36393/session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element/f1e0e771-f9b8-4850-95ba-9dcd2df61f59/text {}
> 2025-05-15 09:30:21,980 [DEBUG] http://localhost:36393 "GET /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element/f1e0e771-f9b8-4850-95ba-9dcd2df61f59/text HTTP/1.1" 200 0
> 2025-05-15 09:30:21,981 [DEBUG] Remote response: status=200 | data={"value":"Mode Eco"} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '20', 'date': 'Thu, 15 May 2025 07:30:21 GMT'})
> 2025-05-15 09:30:21,983 [DEBUG] Finished Request
> 2025-05-15 09:30:21,983 [DEBUG] POST http://localhost:36393/session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element {'using': 'xpath', 'value': "//p[contains(text(), 'Status ')]"}
> 2025-05-15 09:30:22,006 [DEBUG] http://localhost:36393 "POST /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element HTTP/1.1" 200 0
> 2025-05-15 09:30:22,007 [DEBUG] Remote response: status=200 | data={"value":{"element-6066-11e4-a52e-4f735466cecf":"99f005e5-3332-46ee-a155-b726c9b56962"}} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '88', 'date': 'Thu, 15 May 2025 07:30:21 GMT'})
> 2025-05-15 09:30:22,007 [DEBUG] Finished Request
> 2025-05-15 09:30:22,009 [DEBUG] GET http://localhost:36393/session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element/99f005e5-3332-46ee-a155-b726c9b56962/text {}
> 2025-05-15 09:30:22,054 [DEBUG] http://localhost:36393 "GET /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64/element/99f005e5-3332-46ee-a155-b726c9b56962/text HTTP/1.1" 200 0
> 2025-05-15 09:30:22,061 [DEBUG] Remote response: status=200 | data={"value":"Status Charging"} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '27', 'date': 'Thu, 15 May 2025 07:30:21 GMT'})
> 2025-05-15 09:30:22,066 [DEBUG] Finished Request
> 2025-05-15 09:30:22,067 [DEBUG] Response JSON: {"success": true, "mode": "Eco", "status": "Charging"}
[... cut ...]
2025-05-15 09:30:34,453 [DEBUG] http://localhost:36393 "DELETE /session/07837b6d-e3b5-470d-9c92-e3caf4ce7b64 HTTP/1.1" 200 0
2025-05-15 09:30:34,454 [DEBUG] Remote response: status=200 | data={"value":null} | headers=HTTPHeaderDict({'content-type': 'application/json; charset=utf-8', 'cache-control': 'no-cache', 'content-length': '14', 'date': 'Thu, 15 May 2025 07:30:21 GMT'})
2025-05-15 09:30:34,455 [DEBUG] Finished Request
2025-05-15 09:30:34,460 [INFO] 172.30.32.1 - - [15/May/2025 09:30:34] "GET /wallbox/status HTTP/1.1" 200 -
```

derolli1976, 2025-05-15:
Danke für die Logs.
Ich musste von Chromium zu Firefox wechseln, damit das addon auch auf dem Raspi lauffähig ist. Kannst Du mir noch die Logs aus Homeassistant für die Integration selber zur Verfügung stellen?
Grundsätzlich scheint das addon ja zu liefern. Wenn das in HA nicht auftaucht, hängt es irgendwo dazwischen.

Log bitte unter "Einstellungen" / "System" / "Logs" (Homeassistant Core) herunterladen und hier zur Verfügung stellen.

Ich werde das addon aber noch mal überarbeiten und robuster machen. Firefox ist im initalien rendering etwas lahm.

Daddycool2708, 2025-05-15:
Wenn ich auf den Wallbox status mit der aufrufe mit der Adresse http://192.168.178.153/wallbox/status dann dauert es etwa 60 Sekunden bis ein Ergebnis angezeigt wird.

Hier der Log von heute Morgen.

```
2025-05-15 09:24:40.931 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration enpal which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-05-15 09:24:40.934 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration localtuya which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-05-15 09:24:40.937 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-05-15 09:24:40.939 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration enpal_webparser which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-05-15 09:24:52.711 INFO (SyncWorker_3) [fritzconnection] Unable to retrieve resource 'http://192.168.178.22:49000/igddesc.xml' from the device.
2025-05-15 09:24:52.714 INFO (SyncWorker_0) [fritzconnection] Unable to retrieve resource 'http://192.168.178.48:49000/igddesc.xml' from the device.
2025-05-15 09:25:06.151 WARNING (MainThread) [custom_components.enpal_webparser.sensor] [Enpal] Wallbox update failed – no previous data yet: 
2025-05-15 09:25:06.154 ERROR (MainThread) [custom_components.enpal_webparser.sensor] Error fetching Wallbox Status data: Wallbox update failed and no previous data: 
2025-05-15 09:25:24.655 WARNING (zeroconf-ServiceBrowser-_googlecast._tcp-121) [pychromecast.dial] Failed to determine cast type for host <unknown> (<urlopen error _ssl.c:1011: The handshake operation timed out>) (services:{MDNSServiceInfo(name='Pioneer-VSX-932-d6810bcd0ea394cee7351146a5beac00._googlecast._tcp.local.')})
2025-05-15 09:25:25.117 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:25:55.118 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:26:17.153 WARNING (MainThread) [custom_components.enpal_webparser.sensor] [Enpal] Wallbox update failed – no previous data yet: 
2025-05-15 09:26:25.119 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:26:55.119 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:27:25.120 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:27:28.152 WARNING (MainThread) [custom_components.enpal_webparser.sensor] [Enpal] Wallbox update failed – no previous data yet: 
2025-05-15 09:27:55.122 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:28:25.125 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:28:39.151 WARNING (MainThread) [custom_components.enpal_webparser.sensor] [Enpal] Wallbox update failed – no previous data yet: 
2025-05-15 09:28:55.127 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:29:12.154 ERROR (MainThread) [aioharmony.hubconnector_websocket] 192.168.178.30: Response error: No PONG received after 5.0 seconds
2025-05-15 09:29:18.161 ERROR (MainThread) [aioharmony.hubconnector_websocket] 192.168.178.30: Connection timed out for hub 15663504
2025-05-15 09:29:25.128 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:29:50.156 WARNING (MainThread) [custom_components.enpal_webparser.sensor] [Enpal] Wallbox update failed – no previous data yet: 
2025-05-15 09:29:55.129 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:30:25.131 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:30:55.131 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:31:01.152 WARNING (MainThread) [custom_components.enpal_webparser.sensor] [Enpal] Wallbox update failed – no previous data yet: 
2025-05-15 09:31:25.133 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:31:55.135 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:32:12.152 WARNING (MainThread) [custom_components.enpal_webparser.sensor] [Enpal] Wallbox update failed – no previous data yet: 
2025-05-15 09:32:25.138 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 09:32:55.140 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
[... cut ...]
2025-05-15 12:04:44.155 WARNING (MainThread) [custom_components.enpal_webparser.sensor] [Enpal] Wallbox update failed – no previous data yet: 
2025-05-15 12:04:54.539 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 12:05:24.540 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
2025-05-15 12:05:54.540 WARNING (MainThread) [custom_components.enpal_webparser.select] Unknown wallbox mode from sensor: unavailable
```

derolli1976, 2025-05-16:
Danke für das log. Ich sehe gerade, ich habe mich bei der Frage hier mit der IP vertan.

> Nur zur Sicherheit: Kannst Du das bitte mal in deinem Browser einfügen: http://192.168.178.153/wallbox/status
Bitte die IP mal durch die Adresse von Deinem Homeassistant austauschen:

`http://[homeassistant-ip]:36725/wallbox/status` 

Das addon scheint die Daten zu liefern, aber irgendwie kommt die Antwort nicht an. Läuft das addon "normal" unter Homeassistant oder ist da irgendetwas besonders?

Daddycool2708, 2025-05-16:
Also wenn ich auf die Adresse

> http://[homeassistant-ip]:36725/wallbox/status

gehe dann kommt der Fehler "connection refused"

> Wenn Du in der Integration (nicht im Addon) auf "Configure" klickst und dann auf "Submit". Bekommst Du da eine Fehlermeldung?

JA dann bekomme ich den Fehler:

> Die Wallbox-API ist nicht erreichbar. Stelle sicher, dass das Add-on läuft und erreichbar ist! Test: http://homeassistant-IP-ADDRESSE:36725/wallbox/status

Daddycool2708, 2025-05-16:
nach ein paar mal aktualisieren drücken kommt bei http://[homeassistant-ip]:36725/wallbox/status folgendes:

```
{
  "mode": "Solar",
  "status": "NotConnected",
  "success": true
}
```

derolli1976, 2025-05-16:
Also dann sieht es irgendwie so aus, als wenn das Addon bei dir grundsätzlich irgendwelche Netzwerk Probleme hat.
Connection Refused ist sonderbar. Vor allem, wenn es danach plötzlich normal antwortet.
Wie genau ist dein Setup?
Auf welcher Hardware läuft Homeassistant? Bare Metal oder als VM? Welcher Prozessor, wieviel RAM?

Daddycool2708, 2025-05-16:
Es läuft auf einem Synology 918+ NAS als VM und 4GB Ram zugewiesen.

Aber das Wallbox addon version 1.0.5 lief ohne probleme

derolli1976, 2025-05-16:
Ich musste aus Kompatibilitätsgründen das Addon von einer Chromium auf eine Firefox Engine umbauen. Daher benötigt es jetzt offensichtlich mehr Dampf. 
Jedes Addon in Homeassistant ist ein eigener Docker Container. Wie viele andere Addons hast du noch laufen?
Kannst du die testweise einmal deaktivieren?

Ich schaue die Tage aber mal, was ich noch an der Performance machen kann.