derolli1976/enpal #29: Wallbox Parameter werden nicht mehr bereitgestellt

Cuxhavener, 2025-05-17:
Moin. 

Bis gestern lief die Integration einwandfrei. 

Dann abends, ich hatte das Wallbox -Addon geupdated, werden keine Werte mehr für sämtliche Wallbox Entitäten angezeigt (nicht verfügbar). 
Das gleiche aber auch in der Integration von Gittowtf bzw Skiperro. 
Ich vermute Enpal hat da wieder in der Datenbank Änderungen vorgenommen. Ist jetzt Solar Rel 8.46.3-348914 vom 14.5.25. 

Hast du da eine Idee?

derolli1976, 2025-05-17:
Komisch. Bei mir werden die angezeigt. Bin auf der selben Version von Solar Rel.
Was sagt denn die interne Enpal Webseite? Tauchen die da auf? Was sagt das Log in Homeassistant?
Welche genau stehen auf nicht verfügbar?

Cuxhavener, 2025-05-17:
Sind tatsächlich alle Entitäten und auch noch 2 aus der Rubrik Batterie. 

Hier mal das Log

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.enpal_webparser.sensor
Quelle: custom_components/enpal_webparser/sensor.py:176 
Integration: Enpal Solar (Dokumentation, Probleme) 
Erstmals aufgetreten: 16. Mai 2025 um 22:02:34 (9 Vorkommnisse) 
Zuletzt protokolliert: 05:05:20

[Enpal] Error fetching wallbox status:
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/client_reqrep.py", line 1059, in start
    message, payload = await protocol.read()  # type: ignore[union-attr]
                       ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/streams.py", line 672, in read
    await self._waiter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/enpal_webparser/sensor.py", line 176, in async_wallbox_update
    async with session.get(wallbox_url, timeout=15) as resp:
               ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/client.py", line 1425, in __aenter__
    self._resp: _RetType = await self._coro
                           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/client.py", line 730, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.13/site-packages/aiohttp/client_reqrep.py", line 1054, in start
    with self._timer:
         ^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/helpers.py", line 685, in __exit__
    raise asyncio.TimeoutError from exc_val
TimeoutError

derolli1976, 2025-05-17:
Kannst du in der Konfiguration der Integration den Haken für die Wallbox mal raus nehmen bitte?
Fehlen die Entitäten dann immer noch?
Auch gerne noch mal das Log des Addon. Welche Version. Hast du da?
Ab 1.1.0 musst du in der Konfiguration des Addons den Loglevel auf debug stellen.

Cuxhavener, 2025-05-17:
Hallo nochmal. 

Nachdem ich einen Homeassistant Neustart machte, sind alle Wallbox Entitäten wieder da. 
Dafür fehlen 4 aus dem Bereich Batterie. Die sind aber auch im Solar Rel nicht mehr vorhanden. 

Also scheinbar läuft wieder alles so wie es soll. 

Danke nochmal für die Unterstützung. 

[image]

closed 2025-05-17 by derolli1976