derolli1976/enpal #30: [Bug]

Joda1956-JD, 2025-05-19:
### Version der Integration

2.1.1

### Home Assistant Version

aktuell

### Installationsart von Home Assistant

Home Assistant OS

### Was ist das Problem?

Wallbox Sensoren und Buttons nicht erreichbar. Nach Installation, Konfiguration, Start und Aktivierung ändert sich die Anzahl der Entitäten in der Enpal Integration nicht, sie bleibt bei 127. Any hints?

Mit freundlichen Grüßen 
Jörgen Dalhoff

### Schritte zur Reproduktion

Wie oben beschrieben

### Home Assistant Logausgaben

```shell
2025-05-19 18:25:40,673 [INFO] WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:36725
 * Running on http://172.30.33.1:36725
2025-05-19 18:25:40,673 [INFO] Press CTRL+C to quit
2025-05-19 18:26:18,876 [INFO] Retrieving wallbox status...
2025-05-19 18:26:22,716 [INFO] Status retrieved: mode='Fast', status='NotConnected'
2025-05-19 18:26:23,606 [INFO] 172.30.32.1 - - [19/May/2025 18:26:23] "GET /wallbox/status HTTP/1.1" 200 -
```

### Relevanter Konfigurationsauszug

```yaml

```

### Verwendest du das Wallbox Add-on?

Ja

### Logausgaben des Wallbox Add-ons

```shell

```

derolli1976, 2025-05-19:
Nabend. Hast du Home Assistant nach der Aktivierung der Wallbox mal bei gestartet?
Ich benötige bitte die Logs aus Homeassistant für die Integration. Die des Wallbox Addon zeigen, dass dieses zu funktionieren scheint.

Joda1956-JD, 2025-05-19:
Hallo Olli,Das war‘s! Nach dem Neustart sind die Sensoren und Buttons erschienen.DankeMit besten Grüßen und meinem iPad gesendetJörgen

derolli1976, 2025-05-19:
Neustart hat das Problem gelöst.